AdaCore/RESSAC_Use_Case #12: Signalement des inputs desactivés

AnthonyLeonardoGracio, 2017-06-20:
Document: UseCaseDevelopment/Layer1_MMS/RESSAC_muXAV_SystemRQ_MMS_SW_Incr1.doc
Commit: 74aa074

En général, lorsqu’un input est désactivé (e.g : quand il n’y a pas d’USB_Key ou pas de message de la GS), comment le sait-on ? Pareil lorsque les informations ne sont pas encore données (ModeSwitch ou NavigationParameters dans la phase d’init)

ledinot, 2017-06-21:
Yes it is not specified.
The case no key + no GS operator was not anticipated.
No mission data => no mission, F_MM has to reject this case, as for any other kind of problem (failures, boot, viability etc.)

clairedross, 2017-06-22:
Thanks. But how do we detect that an input is not there in MMS? Is there some kind of flag Usb_Key_Connected or Data_From_GS_Received that we can check? Or a default value for these inputs which means 'No_Value'?

clairedross, 2017-06-27:
The cleanest way I can think of is to have some kind of option for the type of the signal, that is, have boolean flag inside the signal to specify if it is present and have the value of the signal being available only when present (a discriminant in Ada). Does it seem suitable to you ?

ledinot, 2017-06-27:
I view it as a software design issue, not as a system specification issue.
The intended functionality is clear (may or may not be available, with possibly ensuing mission cancellation).

I admit that without precise knowledge of the protocol between CP + GS  and MMS, it is difficult to figure out precisely how the NAV parameters will be made available to F_MM, and so how to detect that all or part of them are not available.

This kind of question could also be coined *** for the lessons learnt.

This is  why I insist to consider SYSTEM specification and SOFTWARE specification as tightly coupled.
The “standard view”  first system specification is done with allocation to software, THEN software HLRs can be written and software development can start, is an illusion.
It is OK (sequentiality) for the first 80% of system and software specification .
But then for the remaining 20%, … system-software co-specification (concurrent – collaborative engineering).