AdaCore/RESSAC_Use_Case #13: Changement du NavigationMode pendant le vol

AnthonyLeonardoGracio, 2017-06-20:
Document: UseCaseDevelopment/Layer1_MMS/RESSAC_muXAV_SystemRQ_MMS_SW_Incr1.doc
Commit: 74aa074

Est-ce qu’on peut changer le NavigationMode durant le vol ? Car en cas de conflit sur le mode, il est dit que CP prévaut sur GS (6.9.4.1)

ledinot, 2017-06-21:
Yes, the GS operator can  change the mode, as many times as he wishes.

clairedross, 2017-06-22:
Thanks. If I understand well, GS can change the mode as it wishes as long as CP has set the mode to RP is that is?

ledinot, 2017-06-23:
Yes.
The field operator can prevent GS from taking control over the drone by putting the switch on A position.
In RP position, if GS sends no command, the mission evolves exactly like in A mode.
If GS sets A mode (and drone’s switch on RP position), posterior GS commands are ignored by the drone, except RP command, which sets back the drone to listen to GS commands.

clairedross, 2017-06-23:
Ok, thanks, I understand much better now.

ledinot, 2017-06-23:
Enough to close this issue ? (:-))

clairedross, 2017-06-23:
I guess you can close it if you think that explanations in  6.9.4 are clear enough:
1. In case of conflict on the navigation mode, CP prevails over GS
it was not clear to me that GS could switch the mode to A when CP had said RP...

ledinot, 2017-06-23:
You are right, it is a subtle issue to be clarified.
Because of the mechanical nature of the A/RP switch on CP, there is no way to move it after take-off.

When CP is on RP, it means “GS has authority”.
If GS says “A” in flight, the drone obeys (freeze of Nav’s parameters) but can’t push the switch to sync it with the logical position set at GS level.
So, CP prevails over GS on navigation mode is definitely true for A mode.
For RP mode also, but in a much more disputable sense: “If I (CP) decided that you (GS) has control, then you have it, including on the RP/A switch, you have logical control over it, whatever its mechanical position is.