AdaCore/RESSAC_Use_Case #14: GS et MassPayload

AnthonyLeonardoGracio, 2017-06-20:
Document: UseCaseDevelopment/Layer1_MMS/RESSAC_muXAV_SystemRQ_MMS_SW_Incr1.doc
Commit: 74aa074

«mass of the payload must be given to the system (GS message or CP rotators or USB key)” (6.6.4.1): or dans 4.1.1, on en voit pas comment GS peut founir la masse. De même, dans 4.1.2, USB_Key ne contient pas de masse.

ledinot, 2017-06-21:
Yes it is an error (influence of NAV parameters).
Mass can  only be given by the field operator that loads the payload in the cargo bay, with the 2 rotactors.

clairedross, 2017-06-22:
OK, thanks.

ledinot, 2017-06-27:
It is an error (confusion with the NAV mode/parameters that are double-channeled).
Mass is critical:  only the operator who loads the drone is qualified to give it to the embedded system.
So rotactor only.