AdaCore/RESSAC_Use_Case #3: AV_CP_Switches on listé dans les inputs de F_PT

AnthonyLeonardoGracio, 2017-06-20:
Document:  UseCaseDevelopment/Layer1_MMS/RESSAC_muXAV_SystemRQ_MMS_SW_Incr1.doc
Commit: 74aa074811487 

La sous-fonction F_CM attend AV_CP_Switches en input mais celui-ci n'est pas listé dans les inputs de la sous-fonction englobante F_PT.

ledinot, 2017-06-21:
As a single record, yes. But the information is there, split in several flows:
-ON/OFF_PushButton => Power
- NavigationMode => Mode
- BaySwitch => Bay
- START_PushButton => START
Except Rotactor1{0:9} and Rotactor2{0:9} that were forgotten in F_PT's inputs.

clairedross, 2017-06-22:
Thanks for the precision. We already have PayloadMass in F_PT's inputs, should it be removed if the two rotactors are given explicitely?

ledinot, 2017-06-22:
That’s also an interesting point for the lessons learnt.
It depends on “who” codes the two rotactors?

1.       Is it CP, because it has a small micro-processor? In that case MMS does not need to receive the two rotactors, and my interfaces were right.

2.       Or is CP purely electro-mechanical, and F_PT.F_CM has two receive the two rotactors to compute PayloadMass=10*Rotactor1 + Rotactor2.

I postpone my final answer, but option 1 is the most likely.

clairedross, 2017-06-22:
OK, and then with option 1 F_PT.F_CM still needs to compute the values of each rotactors to store them in its output to GS named CP_Switches ?

ledinot, 2017-06-22:
Yes.
Even with a computerized CP (a smart CP ☺), it would not be in charge of repeating its statuses to GS.
MMS is the control center, and F_CM is in charge of broadcasting datas wherever needed, after elaboration by the control center.
So in option 1, you are right, the rotactors’ values would be “reconstructed” by F_CM since they would not be acquired by analog links  to the rotactors.