AdaCore/RESSAC_Use_Case #5: StartTakeOff et StartLanding non listés en inputs de F_FC

AnthonyLeonardoGracio, 2017-06-20:
Document: UseCaseDevelopment/Layer1_MMS/RESSAC_muXAV_SystemRQ_MMS_SW_Incr1.doc
Commit: 74aa074

F_MM fournit StartTakeOff et StartLanding à F_FC mais F_FC ne les a pas en input

**Réponse initiale**

> Erreur. Il faut les rajouter en entrée de F_FC.

clairedross, 2017-06-28:
Hello Emmanuel,

Just to complement this question, could you also tell us what they are used for? Are they used to start and stop F_FC, that is, it is only running from take_off to start of landing?

Thanks a lot,
Claire

clairedross, 2017-07-05:
Hello Emmanuel,

I think you did not answer my last question, it was probably buried in the mass ;) . However, I am pretty sure F_FC won't stop running during landing, as it obviously needs to brake. So maybe F_FC will start running when Start_TakeOff is True, then do its propulsion and braking during flight, and then starts to do braking only when Start_TakeOff is set to True (or at least CommutationDuration later) ? Then it will stop working when pdot and qdot are both 0?

Thanks,
--
Claire

ledinot, 2017-07-05:
Claire, in your text.

De : Claire Dross [mailto:notifications@github.com]
Envoyé : mercredi 5 juillet 2017 11:41
À : AdaCore/RESSAC_Use_Case
Cc : Ledinot Emmanuel; Assign
Objet : [Message publicitaire : ] Re: [AdaCore/RESSAC_Use_Case] StartTakeOff and StartLanding not listed in F_FC inputs (#5)

Hello Emmanuel,

I think you did not answer my last question, it was probably buried in the mass ;) . However, I am pretty sure F_FC won't stop running during landing, as it obviously needs to brake.

Of course.

So maybe F_FC will start running when Start_TakeOff is True,

Yes.

then do its propulsion and braking during flight, and then starts to do braking only when Start_TakeOff is set to True (or at least CommutationDuration later) ?

Did you intend when “Start_Landing is set to True”?

In that case the answer is yes.

I don’t understand the question about CommutationDuration.

There is a link with activation of braking, but none with understanding the termination logic of F_FC.

Braking is decided by the piece-wise (= gain scheduled) PID controller when the needed torque is negative .

Zero-crossing of the needed torque can be “instantaneous” at software level

(i.e at cycle k  propulsion-torque > 0, at cycle k+1 propulsion-torque < 0, so braking-torque(k+1) > 0 as opposite of the negative propulsion-torque(k+1)).

But the counterpart at physical level cannot be so quick (more precisely: care will have to be taken at scheduling level + timing analysis that the frequency of the task running F_FC is high enough for efficiency  of control but not too high).

This is because the engines and the clutch need some time to stop acting on the shaft.

And one must avoid simultaneous contradictory physical actions on the shared shaft.

CommutationDuration is the “time separator” between opposite actuations on the shared shaft.

The value of this constant has to be tuned by considering the task activation frequency, logical time, real time, the time constants of clutch release, propulsion realse etc.

It plays no role on the termination logic of F_FC.

F_FC terminates when the drone is at rest on ground (or at rest in peace :-)).

Hope this sheds enough light.

E.

ledinot, 2017-07-05:
Claire, in your text.
E.

De : Claire Dross [mailto:notifications@github.com]
Envoyé : mercredi 28 juin 2017 10:14
À : AdaCore/RESSAC_Use_Case
Cc : Ledinot Emmanuel; Assign
Objet : [Message publicitaire : ] Re: [AdaCore/RESSAC_Use_Case] StartTakeOff and StartLanding not listed in F_FC inputs (#5)

Hello Emmanuel,

Just to complement this question, could you also tell us what they are used for? Are they used to start and stop F_FC, that is, it is only running from take_off to start of landing?

F_FC runs from start of mission (so a bit before take-off), up to end of landing, or activation of emergency landing (consequence of Mission_Aborted).

clairedross, 2017-07-06:
OK, thanks for your answers.

-- 
Claire Dross, PhD
Software Engineer
<dross@adacore.com>