AdaCore/RESSAC_Use_Case #17: Domain mesh for tabulated functions in F_MM

clairedross, 2017-06-22:
Hello Emmanuel,
In 6.6.4 Mission Viability Logic, we should use tabulated functions to get the energy level necessary for the mission. Following your answer of #16, it seems that we should also have some kind of matrix giving the center of each cell for these functions, is it the case? And the same holds for GlideDistance probably.
Thanks for your help,
Claire

clairedross, 2017-06-22:
Some additional questions following some answers in #16. Should we use the same Payload_Mass_Grid as in F_FC to compute the energy level? Has icing an impact on viability logic, or said in an other way, should we use for the Estimated_Total_Mass rather than the initial payload mass to look into the viability tables? Thanks again

ledinot, 2017-06-22:
> Should we use the same Payload_Mass_Grid as in F_FC to compute the energy level?

Good question, it was not specified.
So, yes we assume a single payload mass discretization schema, that can simultaneously solve the viability calculations in F_MM, and gain scheduling in F_FC.

> Has icing an impact on viability logic, or said in an other way, should we use for the Estimated_Total_Mass rather than the initial payload mass to look into the viability tables?

No. Icing cannot be foreseen, even today on airplanes with “on-line” advanced meteo services it is very difficult….
So viability computation can only ignore icing risk, and assume non-icing mission.
It applies safety margins however, for possibly  “a little” (flat rate budgeted) icing conditions.
Viability uses payload mass, the only knowable parameter.

clairedross, 2017-06-22:
Thanks, it is very clear. Just a few little questions, to make sure I understand properly. Why do you say  Estimated_Total_Mass (computed by F_FC) is not knowable, because it is estimated? And if F_MM does not use it for its viability logic, why does it use it for?

ledinot, 2017-06-22:
> Why do you say Estimated_Total_Mass (computed by F_FC) is not knowable, because it is estimated?

Yes.
Suppose the engines become less efficient during the mission.
The very coarse estimation algorithm is based on “propulsion effect loss”, and infers ” likely ice additional mass” as soon as there is an estimated efficiency loss.
But it can be because of wearing engines, or because of head wind that slows the drone down compared to the tabulated efficiency records,
without any true contribution of additional ice on the drone.
Nothing is truly knowable on what really happens.

> And if F_MM does not use it for its viability logic, why does it use it for?

Good point.
6.6.4/in-flight phase was written too quickly.

Additional spec needed (TBDL).
The surface response uses the declared payload mass.
But, when the needed energy is compared to the available energy, a safety margin is used (different threshold constants are used).
When EstimatedTotalMass evolves (i.e increases … unless the payload has been droped over the air…:-)), and even more so if it increases quickly,
F_MM applies greater safety margins.

clairedross, 2017-06-22:
OK, thanks for your explanations.

ledinot, 2017-06-22:
You’re welcome.