AdaCore/RESSAC_Use_Case #19: Physical parameters

clairedross, 2017-06-23:
Hi Emmanuel,

in section 4.1.3, three parameters are defined:
p : Drives distance
pdot: Drives speed
q: Drives altitude and pitch
I am not sure I understand very well what they are and how they are used.

From section 6.7.4, I deduce q is an angle (probably the pitch?), but I cannot understand how this angle uniquely determines the altitude of the drone (see section AV's State Vector). Maybe the drone can only fly at a given altitude if it is leaning enough? And then the drone cannot go higher than maybe, 0.5 * L, or it would be too vertical? As for pdot, I suppose it is the speed measured in the direction where the drone is leaning, and then speed is the speed with respect to the ground (still section AV's State Vector)? And then the speed pdot seems to also be uniquely determined by the pitch q (see section Propulsion Control), at least if g is a constant (maybe for g-force), so the drone can only fly at some speed if it is leaning the right way? Then why should there be two measures, one for pdot and one for q? As for p, I don't see what it is used for...

Thanks for your help,
Claire

ledinot, 2017-06-23:
I’m sure you can’t .

This “mysterious” and admittedly incomplete part of the specification is not for the pleasure of bothering implementers.

I don’t want to disclose the actual nature of the dynamics.

In real life it is most often the case: software has to be developed by people who are not given the means to truly understand what they have to implement and test …. !!

So, be sure you didn’t understand, but because of me :-).

“xx drives yyy”  means “xx is used to compute yyy, yyy depends highly on xxx”.

I can give some help, but not as much as you expect.

The AV’s dynamics is 2D:  X axis = Distance(t)=f(p(t)), Y axis = Altitude(t)=g(q(t)).

There is a strong relationship between Altitude(t) and Speed(t).

Without wind disturbances, Speed(t) entirely determines Altitude(t).

Forget about « pitch » and « roll », think the following way: there are constraints on p, pdot, q, qdot, that bound both  the cdg movement (speed and altitude)  and attitude aspects (pseudo-pitch).

And the OEM gave me no additional visibility on that.

clairedross, 2017-06-23:
Ok, fine. Just one last point: Distance does not depend on p, it is directly computed from speed, so what is p used for?

ledinot, 2017-06-23:
As I wrote it, you are right, we could discard p.
I could have written Distance=f(p(t)), and Speed(t)=g(pdot(t)).
I preferred the integration view, but it has the drawback you pointed out.

See it the following way: there is some potential for consolidating Distance(t):
- method1: computed as specified (from pdot)
- method2: computed from p (possibly to be added with a data fusion logic).

clairedross, 2017-06-23:
OK, thanks for your answers.