AdaCore/RESSAC_Use_Case #20: OperatingPoint and ENERGY mode

AnthonyLeonardoGracio, 2017-06-26:
In 6.6.4 (Operating point update management), it's said that F_MM sends the OperatingPoint to F_FC only in ALTITUDE or SPEED mode. And in ENERGY mode, the altitude and speed reference values is computed by F_FC.

In 6.7.4 (Propulsion Control), it's said that when the ENERGY mode is active, then ALTITUDE or SPEED option is selected depending on the two preset altitude and speed values.

The question is: where do the two preset altitude and speed values come from if they don't come from F_MM?

anneperrineporte, 2017-06-27:
I complete the question
in outputs of F_MM we have an operating point 'altitude and speed'
what is F_MM supposed to do there (6.6.4)?
_"The operating point is the reference value F_FC regulates on at a given instant. When it regulates on an altitude value or on a speed value, these values are sent by F_MM to F-FC. "_
Looks like these are data that are managed and used by F_FC

ledinot, 2017-06-27:
Hi Anthony,

> where do the two preset altitude and speed values come from if they don't come from F_MM?

They come from F_MM, in ENERGY mode as well.
I understand that 6.6.4 is a bit misleading:

6.6.4 : « When F_FC is in energy mode, the altitude or the speed reference value is computed by F_FC”
This sentence gives the impression that F_FC becomes independent from F_MM in ENERGY mode, whereas 6.7.4 shows that it is not the case.

6.7.4: “When “Energy” option is active, the “Altitude” or “Speed” option is selected, depending on the respective energy cost induced by the two pre-set Altitude and Speed values.”

Here is the reason of your legitimate concern:  when an energy minimization criterion is used, the SPEED and ALTITUDE parameter values managed by F_MM will no longer be used. The speed and altitude of the AV (not the NAV parameters hold by F_MM)  will result from F_FC’s optimal control algorithms. (6.6.4).

But for this first implementation we cheat, not to embark into complicated things.

The “optimal control” computation of F_FC is: “either  I choose to regulate on Speed (that given by F_MM, i.e the parameter …) or to regulate on Altitude (that given by F_MM as well …).

So ENERGY  for this first implementation is only the choice by F_FC between mode ALTITUDE  and mode SPEED,

but in both cases the choice it makes is a the preset value hold by F_MM (the NAV parameter).

Confusion (induced by my writing between speed=AV’s state variable, and speed = NAV parameter (resp. for altitude)).

Don’t bother with ENERGY mode. It is a provision. Any quick hack is OK.

Emmanuel

ledinot, 2017-06-27:
Ø  Looks like these are data that are managed and used by F_FC.
Exactly.
F_MM controls the value of the navigation objectives (distance, cruise speed, or cruise altitude),while  F_FC takes the objectives and is in charge of fulfilling them by appropriate torque commands over time, whatever perturbations may occur.