AdaCore/RESSAC_Use_Case #22: Switching between Climb, Cruise, and Descent modes in F_MM

clairedross, 2017-06-27:
Hello Emmanuel,

I was looking at F_MM, and I was wondering how it could choose whether it was in Climb, Cruise, or Descent mode. The only information I could find is in 6.6.4 (4. Cruise), where it is said that, when the navigation parameters are updated, or whenever the operating point is changed, a transient climb or descent phase is activated to capture the new operating point. I cannot find more precise information. Does F_MM use the Current_Altitude/Current_Speed given by F_FC to try and guess where it is? It is not so obvious when on ENERGY mode... Or does it stay in a single Descent_Climb mode (that we could call Transient instead) only when the operating point has just been updated, the time for F_FC to capture a new operating point?

Thanks for your help,
Claire

anneperrineporte, 2017-06-27:
Hi Emmanuel, Claire

We discussed that with Emmanuel: the Climb, Cruise, Descent status should come from F_FC. 

@ledinot do you confirm?

clairedross, 2017-06-27:
OK, but then there will be some time lapse between the change in the operating point detected by F_MM and the change in the Climb, Cruise, Descent status, as it would require F_MM to send the new operating point to F_FC before getting back the status. Is it still OK with 6.6.4 (4. Cruise)?

ledinot, 2017-06-27:
Does F_MM use the Current_Altitude/Current_Speed given by F_FC to try and guess where it is?

It is the other way around : F_FC gets the NAV parameters (i.e the NAV objectives) from F_MM, and compares to the current state of the drone.

If there is a significant difference (error value used by the PID algorithms) F_FC  starts tracking the new objectives (reference trajectory generation) because there is no longer equilibrium (= cruise) between the desired and the achieved.

It is not so obvious when on ENERGY mode...

Addressed in another issue.

Cdlt,

E.

.

ledinot, 2017-06-27:
Yes I confirm.
I explained to Sylvan and during the last RESSAC workshop that what make a transition between two phases is that the operating point has been set to value significantly different from the current operating state of the drone.
If the set point is higher than the current state, a climb phase starts (resp. lower/descent).
Since F_FC computes the current state of the drone, i.e (distance(t), speed(t),altitude(t)), it the place where phases and transients are known. Cruise = equilibrium.

ledinot, 2017-06-27:
Yes it is compatible.
The latency of data transfer between F_MM and F_FC is expected to be very low (intra software communication, message passing possibly, but no transfer to external persistent memory or to network.
So no “loss of time” likely to impede flight control or navigation.

clairedross, 2017-06-28:
Ok, thanks for your answers. Then I will just assume the presence of a signal from F_FC to F_MM which returns the current phase (CLIMB, CRUISE, or DESCENT) and use it to condition this state in F_MM.

clairedross, 2017-06-28:
For what it is worth, 6.6.4 (4. Cruise) seems more of a functional behaviour of F_FC than of F_MM to me...