AdaCore/RESSAC_Use_Case #24: How does F_MM know the drone has landed?

clairedross, 2017-06-28:
Hello Emmanuel,

In 6.6.4 Figure 3, we see that Landing terminates when pdot=qdot=0. But, since F_MM does not have pdot as an input nor q (let alone qdot which can only be estimated), I don't understand how F_MM knows that the drone is landed.

Thanks for your help,
Claire

ledinot, 2017-06-29:
Hi Claire,

I confirm F_MM should notre have access to p pdot  q qdot  because it  is not Mission/Nav  data.

So  if she needs to diagnose landing has occured, she can do it by means of Speed = 0, Altitude = 0.
Consider I gave a wrong criterion, apply this new  MM-level one.

E.

-------- Message d'origine --------

clairedross, 2017-06-30:
Hi Emmanuel,
Thanks for your answer, this is clearer now.
Claire

ledinot, 2017-07-05:
Yes.
In a previous answer, the following specification fix was proposed:
F_MM uses Altitude =0 and Speed=0  to detect end of landing, instead of pdot=qdot=0.