AdaCore/RESSAC_Use_Case #25: Clarification on unit of pdot

AnthonyLeonardoGracio, 2017-06-28:
Document: UseCaseDevelopment/Layer1_MMS/RESSAC_muXAV_SystemRQ_MMS_SW_Incr1.doc
Commit: 74aa074

In 4.1.3, it's said that pdot drives the speed of the AV. But in the Guarantees listed in 6.7.3.2, pdot is compared with parameters that have Angle.s-1 for unit.

Is pdot an angular speed then? I initially thought that it was driving the speed of the AV relatively to the ground.

ledinot, 2017-06-29:
To feel uncomfortable here is legitimate.
pdot  actuaĺly is an angular speed.
Don't try to interpret it, or see it as the propeller shaft speed.
Translational speed of the drone can be computed from pdot plus q and paramters.

-------- Message d'origine --------

AnthonyLeonardoGracio, 2017-06-29:
Ok, thanks for the answer. I am closing the issue then.

closed 2017-06-29 by AnthonyLeonardoGracio