AdaCore/RESSAC_Use_Case #27: Getting the NavigationOption from CP

clairedross, 2017-06-29:
Hello Emmanuel,

While looking at the inputs of MMS (4.1), I do not see how we can retrieve the NavigationOption (SPEED, ALTITUDE, or ENERGY) when in A mode. Is there a default value for this option, like we always are in ENERGY mode in that case, or should this input be supplied by CP in some way (additional switch or additional information on the USB key)?

Thanks for your help,
Claire

clairedross, 2017-07-12:
Hello Emmanuel,

Do you have an answer on this issue?

Thanks for your help,
Claire

ledinot, 2017-07-12:
Sorry Claire for having forgotten #27.
It is not a default option, it is provided by the CP, and more precisely by the parameters stored on the USB key.
Cdlt,
E.

clairedross, 2017-07-12:
Ok, thanks.