AdaCore/RESSAC_Use_Case #32: Sub Functions Activation Logic (F_EL, F_PT, MMS)

SylvanDissoubray, 2017-08-30:
§5 says:
Both F_PT and F_EL are activated after MMS set-up.
(This is strange wording since MMS set up is an activity of F_PT itself (F_MM), so it cannot be activated after itself)

§6.5.2 says:
F_EL plays the role of a FDIR mechanism wt. F_PT. Both functions are activated after power-on.

I suggest keeping only the correct one (seems to be the second one)
Keeping it in 6.5.2 seems the correct scope.

ledinot, 2017-09-04:
(This is strange wording since MMS set up is an activity of F_PT itself (F_MM), so it cannot be activated after itself).

Set-up in this sentence is not “mission set-up”, addressed by F_MM and part of F_PT.
It is “multi-system set-up” after power-on at CP, it is the booting sequence (equivalent on a PC to BIOS, MBR, launch of the OS etc.).
In increment 1, multi-system = MMS = one computer without HW …, so no specification aspects, but this second kind of set-up is really an issue for increment 2 and after.

Will be made more precise in §5.

SylvanDissoubray, 2017-09-04:
Fine for me !
May be the multi system set up could be called something like "power on set-up" in contrast with mission set-up