AdaCore/RESSAC_Use_Case #33: Interfaces of F_CM (more specifically with F_MM)

SylvanDissoubray, 2017-08-31:
§6.6.2 says 
For mission preparation, there are redundant input channels: GS messages and CP devices (USB key, rotators and switches). The communication function (F_CM) manages the possible inconsistencies between the redundant channels. F_MM receives mission parameters consolidated by F_CM. 

§6.6.2.1 Input interface of F_MM receives redundant channels from F_CM, USB from F_CM

§6.9.1 
1.	It manages possible inconsistencies between GS-supplied and AV-supplied navigation parameters,
2.	It dispatches the software inputs to the functions,
==> Suggests to me F_CM performs input *selection* (priority logic and filtering, dispatching only selected inputs

§6.9.2 interfaces of F_CM is inconsistent ? (outputs to F_MM not listed ?)

§6.9.4 This paragraph uses the word *selection*, it seems §6.6.2.1 does receive too many inputs....

Note, fig 2 does not seem accurate with this respect.

ledinot, 2017-09-04:
§6.6.2 says
For mission preparation, there are redundant input channels: GS messages and CP devices (USB key, rotators and switches). The communication function (F_CM) manages the possible inconsistencies between the redundant channels. F_MM receives mission parameters consolidated by F_CM.
I made a mistake when writing this.

F_CM has no “applicative knowledge” to do so appropriately in all cases, indeed.
Question: I’m wondering whether the point has already been raised by AdaCore (no time to check at present).

General rule: consolidation/fusion  of redundant channels/datas are made by the “client-functions” (F_PT’s subfunctions, F_EL).
F_CM only collects and dispatches the multiple instances to the “client-functions”.

SylvanDissoubray, 2017-09-04:
OK for me.

(I tried to check preceeding discussion threads to avoid repetition, but number and length of some threads makes compilation of required changes tricky. Sorry for repeated question if any).

clairedross, 2017-09-05:
On 04/09/2017 16:08, Emmanuel Ledinot wrote:
> F_CM has no “applicative knowledge” to do so appropriately in all 
> cases, indeed.
> Question: I’m wondering whether the point has already been raised by 
> AdaCore (no time to check at present). 

It has been raised already:

https://github.com/AdaCore/RESSAC_Use_Case/issues/4

-- 
Claire Dross, PhD
Software Engineer
<dross@adacore.com>

SylvanDissoubray, 2017-09-05:
I had not seen page 2 of issues !!!
Sorry !